Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #1: Not able to compile the book

nitin5, 2015-01-07:
Hi I am trying to compile this book in texWorks in Ubuntu.
I am getting the error that /begin{document} is not found which obviously is present!!
Any help will be appreciatied

Rgds
Nitin

apnorton, 2015-01-11:
I'm on Windows 7 using TeXMaker and am not running into this problem when compiling.  (Just to confirm it isn't a problem with the files, but probably something with your TeX configuration.)

If you're simply wanting to compile the book because you want to read the PDF (and you aren't trying to edit it or anything), you can download a (not bleeding-edge, but up-to-date) PDF from the [releases](https://github.com/correll/Introduction-to-Autonomous-Robots/releases) page.  (I mention the release page because it took me a while to find it, myself.)

nitin5, 2015-01-11:
Thank you Andrew
I agree that i want to read df version of book :)
Thanks for the link.

I will relook into tex configuration. I am able to compile papers but not
the book.
Rgds
Nitin

closed 2016-02-28 by correll