Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #20: Bibtex string for proper referencing

diesbot, 2017-05-15:
Greetings.
First off - great work with this book and thank you for sharing it under the Creative Commons Licence.

For those who want to reference this book, it would be very helpful if a full bibtex-string is provided in the readme-file of this project.
Currently the only information I can find is this old string via Google Scholar.

```
@misc{correll2014introduction,
  title={Introduction to Autonomous Robots},
  author={Correll, Nikolaus},
  year={2014},
  publisher={CreateSpace Independent Publishing Platform}
}
```
It would be very nice, if you could provide an up-to-date reference.

Thank you and keep up with the nice work.

correll, 2017-05-16:
Thanks for your recommendation. I have updated a bibtex string.

diesbot, 2017-05-17:
Thank you very much.

closed 2017-05-17 by diesbot