Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #22: Jacobian in equation 8.12 not clear

choffmann, 2019-10-22:
Hi, I don't understand how you reached the Jacobian in eq. 8.12. For example the very first element in this matrix: Here it seems you only derive ds by dsr which is indeed 1/2. But what about the cos term? dtheta is also a function of dsr, isn't it? However, it seemed you regarded it as a constant in respect to dsr?

jlblancoc, 2019-10-23:
Use eq. (8.8) into (8.5), then take partial derivates wrt to $\Delta s_r$, and here's were the 1/2 appears. Then the cos() remains untouched since the Jacobian is, by definition, a matrix of partial derivatives, so everything else that is not the variable wrt you are taking derivatives, is a constant.

choffmann, 2019-10-23:
However if you look closely the cos term is actually **not** constant wrt s_r or s_l right? dtheta is a function of s_r or s_l, or am I missing something?

jlblancoc, 2019-10-23:
In partial derivatives, you should NOT take into account those
dependencies, just what variables directly appear in the expression...

Anyway, let's see what the author says :-)

correll, 2019-10-23:
I think choffman is right, \Delta \theta is definitely also a function of \Delta s_r and the product rule should apply also for partial derivatives, adding an additional term (-1/8sin(...)). I will look more into that why I made that choice at the time. Thanks A LOT for taking the time to reading, checking the equations, and contributing!

correll, 2019-10-24:
I did some more reading, and the Jacobian is definitely missing terms that should come from the product and chain rules. I will update this soon or accept pull requests. I also noticed that equation 8.7 misses the axle length in the denominator. Thanks again for pointing this out.

choffmann, 2019-10-25:
@correll Ok, I will be at ROScon and IROS the next 2 weeks. When I'm back, I'll have a look again and hopefully come up with a correct Jacobian. ;)
PS: It would speed up the process if you could list your readings.

correll, 2019-12-31:
I did not get to this yet. The derivative is only half of the product rule, the other term is still missing. The equation numbering might have changed since I added a chapter.

closed 2020-03-07 by correll