Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #33: Path planning: obstacles are grown by *half* of the longest extension from the origin?

goekce, 2020-12-16:
Dear Nikolaus,

from path planning chapter:

> ... all the obstacles in the environment are grown by half of the longest extension of the robot from its center.

Shouldn't we grow the obstacles simply by the full length of the longest extension from robot's *origin*?

[image]

Here is another example:

http://www.cs.columbia.edu/~allen/F17/NOTES/lozanogrown.pdf#page=6

correll, 2020-12-16:
I don’t really understand the approach they are using. It also seems they don’t grow the object into every direction. Maybe it is computationally simpler than my approach when considering vector objects. 

I drew over their example to explain my approach.

I think at the end its just about the concept of C-space and growing a grid map can be done with a dilation operation.

Best,
Nikolaus

goekce, 2020-12-16:
> I don’t really understand the approach they are using. It also seems they don’t grow the object into every direction. Maybe it is computationally simpler than my approach when considering vector objects.

I think they assume that the robot (object) never rotates.

> I drew over their example to explain my approach.

I cannot see any drawing.

> I think at the end its just about the concept of C-space and growing a grid map can be done with a dilation operation.

Makes sense.

Actually my point was more about the wording in your example:

> are grown by *half* of the longest extension of the robot *from its center*.

Shouldn't we rather say: In the figure the obstacles are grown by the *longest extension from its center*.

correll, 2020-12-16:
I agree on the wording!

PR #34 (2020-12-28): typo: `longest extension`, not `half of the longest extension`
closes #33
closed 2020-12-28 by correll