Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #35: Save all SVG images as PDF

correll, 2020-12-30:


correll, 2020-12-30:
Make sure page size is adjusted to content

alecive, 2020-12-30:
This https://castel.dev/post/lecture-notes-2/  is a pretty good tutorial on how to convert to tkiz

alecive, 2021-01-04:
Please see here for my first test: https://github.com/correll/Introduction-to-Autonomous-Robots/commit/34d1659cebb67d5afd27ee5d2c7ec6f563c54861

It seems to be working great!

alecive, 2021-01-04:
For the record: please remember to force-add the PDF file of the figure!

alecive, 2021-01-04:
See this commit https://github.com/correll/Introduction-to-Autonomous-Robots/commit/f7c88e2f9ff636e6fdbca4509b0d0b0cdcfc8ba2 for a way to add equations to the inkscape fig.

alecive, 2021-01-04:
I finished converting all pictures in the kinematics chapter. It was a drag, but it looks great!

alecive, 2021-04-28:
Update: finished with the kinematics chapter with https://github.com/Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots/commit/8c139c4b930e278056ad98bd03478d8e88bec6e0

alecive, 2021-05-18:
@correll @crheckman @hayesbh I gave all the chapters a pass. All figures are converted to PDF except from those in https://github.com/Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots/issues/60. If you create new figures, please export them to PDF directly using the tutorial above. I'm closing this for now.

closed 2021-05-18 by alecive