Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #37: Table of Contents

alecive, 2020-12-30:
This is the current ToC:

1. Introduction
2. Locomotion and Manipulation
3. Forward and Inverse Kinematics
4. Statics and Force Considerations
5. Path Planning
6. Task execution
7. Sensors
8. Vision
9. Feature extraction
10. Artificial Neural Networks
11. Uncertainty and Error Propagation
12. Localization
13. Mapping
14. Simultaneous Localization and Mapping
15. Grasping
16. A Trigonometry
17. B Linear Algebra
18. C Statistics
19. D How to write a research paper
20. E Sample curricula

Ideas for improvement/current issues:

- [ ] Grasping is kinda hanging in there -- I recommend placing it after the statics chapter
- [ ] Sensors -- where to place it?
- [ ] ANN -- where to place it?

correll, 2020-12-30:
How about splitting the book into three parts:

- Actuation
- Sensing
- Computation

For this to work, we should move path planning and task execution into the computation part and split the "grasping" chapter into two.

alecive, 2020-12-30:
That seems nice!

correll, 2020-12-30:
The TOC is now organized by the groups actuation, sensing, and computation. Planning and task execution have been moved toward the end of the book (after localization). Grasping is split into "grasping" and "manipulation".

closed 2020-12-30 by correll