Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #40: Meta-chapters (Actuation, Sensing, Computation)

alecive, 2021-01-04:
With reference to this https://github.com/correll/Introduction-to-Autonomous-Robots/commit/3b83c5b6960ff676056d848be71dc6f08743bfdb , shall we actually have meta-chapters (Actuation, Sensing, Computation) to wrap around our chapters? Because I don't see these meta chapters in the commit above.

correll, 2021-01-04:
I believe that starting with the mechanism remains important to motivate why we actually need sensors, computation etc. I plan to implement the following structure

- Introduction
- Part 1: Mechanisms
- Part 2: Sensors
- Part 3: Actuation
- Part 4: Computation

Maybe we can move uncertainty into "sensors"

alecive, 2021-01-04:
FYI right now the code does not compile any more as there is no definition of `\part`

crheckman, 2021-01-05:
This should now be resolved with 96dcdead767f6675991324a4644c639ce7330b79.

closed 2021-01-05 by crheckman