Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #41: Move Section "3.1. Coordinate Systems and Frames of Reference" to appendix

alecive, 2021-01-04:
As per title: I think it should be considered as a tool/notion that students should have rather than a thing they are learning from the book. I recommend putting it after "linear algebra" and before "statistics". Some minor changes to chapter 3 should be done and I'm happy to do them!

correll, 2021-01-04:
I believe this is the first time the students will hear about these concepts in a CS curriculum, don't you think?

alecive, 2021-01-04:
Yes, but can't we refer to the chapter and point them there rather than it occupying most of it? It would make for a slimmer chapter (plus, you could use it beyond the IK chapter)

correll, 2021-01-04:
The book has many points where a introductory CS lecture has to rely on the Appendix to make sure all the students are on the same page. I agree that "random variable to sensor noise" is in the same category as "coordinate systems to forward kinematics" in that they link foundations from different fields to robotics. I believe that the former is part of years one and two in a CS curriculum, but coordinate systems are not. Maybe we can shelf this one and reassess as the content in the appendix grows?

crheckman, 2021-01-05:
Many robotics textbooks take one of two approaches on these sorts of abstract topics (uncommon to CS, broadly applicable in many areas for robotics):

1. have a freestanding chapter to discuss the topic (useful for showing the content as important), or
2. add an appendix for the topic (makes it easy on the non-CS students who likely have already seen it)

What is being suggested by including it in another chapter is different:

3. introduce the topic while introducing the motivating use-case.

There are benefits to this approach: it makes it clear why we're learning the topic, and the narrative is very direct, almost conversational. However, it makes referencing to this universal topic across chapters somewhat problematic. Since it's integrated with the material, it's neither a) very easily referenced nor b) standalone, since it's so tightly tied to the narrative of the motivating use-case.

Based on the tone we've all agreed to adopt for this book, my vote would be to live with complications a) and b) above in order to maintain that tone.

correll, 2021-01-05:
>    1. introduce the topic while introducing the motivating use-case.

Thanks, Chris, for articulating the challenges with this approach. I agree,
these are exactly the advantages and drawbacks of the book and I'd also be
in favor of sticking to the conversational style. A solution to this issue
could be to introduce some redundancy, a contextual introduction of the
topic in the main narrative of the book and a condensed appendix version.

alecive, 2021-01-06:
So what we have right now is 3): the Coordinate Systems section is within the  Kinematics chapter because that's where it is needed.
What I was recommending was 2), i.e. to move it to the appendix. But I'm fine with either solution! Justlet me know

crheckman, 2021-01-06:
I think we'll constantly find ourselves in the place where we need to make tricky organizational decisions to maintain the conversational tone. However I concur with @correll: try to hem close to the the style of 3. (introducing abstract topics in conjunction with their motivating use-case) when it is appropriate, and make meta-decisions about where it's _most_ appropriate. Here I think Coordinate Systems in the Kinematics chapter is appropriate.

This isn't really an "issue" so maybe best to close it (leaving that to @alecive) but it's good to keep for posterity I suppose.

alecive, 2021-01-06:
Sounds good! For the record: an issue can also be an avenue to discuss "strategy" and record decisions for future reference :-)

closed 2021-01-06 by alecive