Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #42: Split uncertainty and error propagation

correll, 2021-01-04:
Given the new multi-part organization of the book, I believe the sensors part should conclude with introducing the concept of uncertainty, whereas error propagation should be introduced only in the computation part. Somebody needs to split the uncertainty chapter accordingly, possibly @crheckman who is currently working on this topic.

crheckman, 2021-01-04:
I could potentially do this, but actuators/motion don't currently introduce uncertainty when it is being introduced, so the argument could be made that it should actually be introduced there?

correll, 2021-01-04:
There is no actuators chapter right now. It will be after sensors and we
should refer that actuators also introduce uncertainty. The current
mechanisms chapter offers an opportunity to introduce uncertainty at two
parts:

1) Motion - wheel slip
2) Grasping - slip/failure

I think we should mention these things, but refer to further chapters for
how to treat them. I think they are too complicated/non-linear to serve as
good examples to introduce the topic. A sensor with Gaussian noise is
perfect instead.

crheckman, 2021-04-16:
Old issue irrelevant to current structure; closing.

closed 2021-04-16 by crheckman