Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #44: Revised Kinematics chapter

alecive, 2021-01-06:
@correll I have finished revising the kinematics chapter. I would say I did not do any major change, but I still did some changes you should double check:

- [ ] I left many TODO items in red throughout the text for you to take a look. Please do and either fix them or let me know what I should do
- [ ] I tried to add a little bit more structure to the whole chapter, with more cross references so as to better appear as a coherent story. In particular, I modified and enriched the beginning of both the FK and the IK chapters to refer to each other
- [ ] I implemented a more extended set of changes into the IK chapter. Students have had issues with it in the past and I hope to have improved it!

correll, 2021-01-09:
I have processed some of the todos, but have questions on the following:

-Figure 3.6: When I apply the right-hand rule the z-axis comes out of the page. Therefore, the positive rotation is counter-clockwise. Do you think \theta_I should rotate clockwise? 

- "The speed of the robot along its y-axis is always zero". You say "that there is such a thing can be seen immediately when imaging the robot's wheels spinning in opposite directions". That there is what thing? 

- Section 3.2: Yes, I think a figure showing how the transforms can be concatenated along a kinematic chain would be really helpful.

alecive, 2021-01-14:
1. Figure 3.6 : we had some confusion in the past with this picture and it stems from the fact that one usually has the angle "starting" from the X axis. In that case, do we concur that {R} (and therefore theta_I) is rotated approximately -20 degrees with respect to {I}? If we agree on this, then theta_I should point downward exactly because positive direction is the counter-clockwise direction. By pointing clockwise, we clarify that it's a negative angle.
2. "that there is such a thing can be seen immediately when imaging the robot's wheels spinning in opposite directions" This was your sentence! And I don't understand what you meant
3. ok

correll, 2021-01-14:
To 1: Yes, I agree theta_I is misleading as X_I is at -20 degree from the robot's perspective. As the z-axis comes out of the figure in both cases, the positive rotation is counter-clockwise. It might be therefore more clear to label the angle with -\theta_i. 

To 2: I get it. I thought the sentence is part of your question. It refers to the existence of the z-axis. Albeit the robot does not move up and down, there is a z-axis that the robot rotates around. I will clarify. Thanks.

commit da774310 (2021-01-14): close #44
closed 2021-01-14 by correll