Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #46: Minor revision of the Kinematics chapter

alecive, 2021-01-08:
I think that right now the last subsection in the kinematics chapter is confusing. It is not clear why there is an " Inverse Kinematics using Feedback Control" at the same level as "Inverse Kinematics", nor why it mentions control concepts within a Kinematics chapter. Furthermore, the Inverse Jacobian technique (which could be just called Differential Kinematics ) is again under " Inverse Kinematics using Feedback Control" which is kinda strange.

@correll I understand your perspective, which is "I don't want to overwhelm students with jacobians and stuff", butI need to refer to jacobians a lot in the force chapter, so we can't get away from it in the first place.

correll, 2021-01-08:
I'm thinking about this as a numerical solution to the IK problem. There are two perspectives on this, a controls one where the robot is moving along the trajectory that is being computed and a purely computational one in which iteration is used to find a solution, then the robot is sent there. 

I'm all for breaking out the inverse Jacobian under its own heading, but the idea is to explain the concept along with an application.

alecive, 2021-01-08:
Got it--I'll think about it a bit more.

correll, 2021-01-08:
One solution could be to simply rename the subsection to "Differential Kinematics" and elevate to "section", then explain that often we are interested in the relationship between joint and end-effector speeds to introduce the idea of the Jacobian. We can then have a subsection there that introduces the idea of using the Jacobian as part of a numerical solution. 

I notice that we don't have a single chapter on "feedback control". We could limit this to PID control with manual tuning using Ziegler-Nichols. Everything else would require a lot more background to introduce stability etc.

alecive, 2021-01-08:
> One solution could be to simply rename the subsection to "Differential Kinematics" and elevate to "section", then explain that often we are interested in the relationship between joint and [...]
 
 This is what I was thinking too
 
 >  I notice that we don't have a single chapter on "feedback control". We could limit this to PID control with manual tuning using Ziegler-Nichols. Everything else would require a lot more background to introduce stability etc.
 
 I thought about that. I think it's nice not to have too much control as it scares undergrads away. We should have __some__ though, I guess

alecive, 2021-01-15:
Ok I had another big pass and I would say now the whole chapter is 60% changed from before. Added new diff kinematics section, improved clarity, moved some things around. Please check it out!

I still have to add redundancy. After that one, I won't ever touch this chapter again 😎

closed 2021-05-26 by correll