Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #47: Actuator chapter added

correll, 2021-01-12:
I added a preliminary chapter on actuation. Can you please all proof-read and content as you see fit?

alecive, 2021-01-15:
Revised -- fixed some typos here and there and left a todo for you to fix. 

I think that there could be two beneficial additions:
1. the figure -- as you said--as right now it's a full wall of text and it is difficult to keep focused. I imagine students having a lot of confusion in their head to try to remember all these motors and their differences
2. something about transmission and gear reduction at the end? Just to have students know that it exists

correll, 2021-01-15:
I envision two figures, one showing a simple AC motor and commutator for
Dc, the other showing a brushless stepper and control signal. If you want
to add a section about reductions, we can add another picture there. The
gear section should also explain the non-trivial harmonic drive and issues
around backdrivability that will impact safety.

alecive, 2021-01-15:
Revised -- fixed some typos here and there and left a todo for you to fix. 

I think that there could be two beneficial additions:
1. Having a figure -- as you said. Right now it's a full wall of text and it is difficult to keep focused. I imagine students having a lot of confusion in their head to try to remember all these motors and their differences
2. Section 8.1.7 is a bit rushed and you should revise it. I don't know much about this stuff so I don't know how to fix it myself.
3. I added this to 8.2.1: ```However, they are costly to maintain, difficult to control, and they are usually characterized by a low bandwidth: that is, they will never be as reactive as electric motors, and they might be infeasible in human-populated environments where speed of reaction is paramount for safety considerations (\cref{sec:actuators:safety}).```

closed 2021-05-26 by correll