Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #48: Vision chapter needs to be better integrated into sensing

correll, 2021-01-13:
The Vision chapter is at an odd place between sensing and computation. If it stays in the sensors part, we should modify language slightly and make clear that all of the computation in this chapter falls into the category "signal processing". 

I believe that makes sense as a lot of the sensors, e.g. Lidar, Stereo vision etc, that we describe have a significant computational component that is usually shielded from the roboticist. A related idea is to rename that part "Sensing and signal processing".

crheckman, 2021-04-16:
I will handle.

alecive, 2021-04-27:
Reopening: you will handle or you did handle? 😎

crheckman, 2021-04-27:
Having two to-do lists is redundant and there's already #51, so I consolidated.

closed 2021-04-27 by crheckman