Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #49: Sensing and actuation together?

alecive, 2021-01-14:
I propose having a single "SEnsing and Actuation" part rather than two separate parts. It would be a better balance with the remaining parts in terms of chapters, and would not leave the actuators chapter all alone :-)

correll, 2021-01-14:
> to @correll <https://github.com/correll>.

This is the smartest idea I have heard in a long time!

closed 2021-01-14 by correll

alecive, 2021-01-15:
Haha glad you liked it! Then, after the actuators there could be a small chapter on PIDs and low-level motor control.

correll, 2021-01-15:
We should think more about signal processing and controls vs. computation.
The Webots manual is actually quite good at explaining how Webots
implements PID position and velocity controllers.

commit c36d4af5 (2021-01-14): close #49