Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #56: Improved figure 3.7

alecive, 2021-04-27:
@correll  I have changed how the forward kinematics of a differential wheel robot is presented. This is the single most painful element of the whole intro to robotics course, and I think it's now easier to understand. 

Here are the two pictures side by side for your information:

[image]

Not much to do; just an acknowledgment to get feedback.

alecive, 2021-04-27:
I will obviously remove the one on the top -- I just wanted to show the comparison

correll, 2021-04-27:
Looks great, what is there left to do in the assignment?

alecive, 2021-04-27:
Just wanted you to see it :-) Closing!

closed 2021-04-27 by alecive