Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots #72: Is there any way to get the very last version?

thephet, 2024-02-13:
Hi there,

I have the version from December 21, which is the one you have in releases. But I see you have been doing updates, even as last week based on the commits.

I wonder if it would be possible to get the very last PDF.

correll, 2024-02-16:
> <Introduction-to-Autonomous-Robots/Introduction-to-Autonomous-Robots/issues/72
> @github.com>

Hi Juan,

You will need to compile the PDF yourself due to our copyright agreement.

Best,
Nikolaus

Nikolaus Correll, Professor
IRT Multi-functional Materials <http://mfm.colorado.edu>
correll.cs.colorado.edu

alecive, 2024-02-16:
Do we have compilation instructions on the README.md file? Maybe we should add them.

lrwham, 2024-08-25:
Downloading the tools is easy, but takes longer than compiling the book. The suggestion to use Overleaf is reasonable for most.

For the rest, install Texlive [https://www.tug.org/texlive/](https://www.tug.org/texlive/)

Then the one liner below builds the book from the root of the repository.
`pdflatex -pdf book.tex`

If you're on MacOS, the MacTeX package from Brew works [https://formulae.brew.sh/cask/mactex](https://formulae.brew.sh/cask/mactex).
It is a large slow install unfortunately.

On Debian derived distros try `apt install texlive texlive-bibtex-extra texlive-latex-extra`

laventura, 2025-09-17:
Added instructions on how to compile the PDF using `pdflatex` and verified it on MacOS. 
See `HOW-TO-COMPILE.md` in the root directory of the repo.